Matheus-Garbelini/sweyntooth_bluetooth_low_energy_attacks #4: Compatible with Adafruit feather nrf52?

keisentraut, 2020-03-11:
Hi,
thanks for this research! I'm currently trying to reproduce it, but only have an [Adafruit feather nrf52](https://www.adafruit.com/product/3406) at hand. It has the same chip, but is connected with an USB-UART adapter and therefore shows up as `/dev/ttyUSB0` (instead of `/dev/ttyACM0` ). 

When I try to flash it, I get the following error:
```
# adafruit-nrfutil dfu usb-serial -p /dev/ttyUSB0 -pkg nRF52_driver_firmware.zip 
2020-03-11 16:42:06,065 No trigger interface found for device with serial number: 018C5CD3, Product ID: 0xEA60 and Vendor ID: 0x10C4

Traceback (most recent call last):
  File "/bin/adafruit-nrfutil", line 11, in <module>
    load_entry_point('adafruit-nrfutil==0.5.3.post12', 'console_scripts', 'adafruit-nrfutil')()
  File "/usr/lib/python3.8/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3.8/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3.8/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3.8/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3.8/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3.8/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/lib/python3.8/site-packages/nordicsemi/__main__.py", line 993, in usb_serial
    do_serial(package, port, connect_delay, flow_control, packet_receipt_notification, baud_rate, serial_number, False,
  File "/usr/lib/python3.8/site-packages/nordicsemi/__main__.py", line 949, in do_serial
    dfu.dfu_send_images()
  File "/usr/lib/python3.8/site-packages/nordicsemi/dfu/dfu.py", line 119, in dfu_send_images
    self._dfu_send_image(self.manifest.softdevice)
  File "/usr/lib/python3.8/site-packages/nordicsemi/dfu/dfu.py", line 88, in _dfu_send_image
    self.dfu_transport.open()
  File "/usr/lib/python3.8/site-packages/nordicsemi/dfu/dfu_transport_serial.py", line 217, in open
    self.__get_mtu()
  File "/usr/lib/python3.8/site-packages/nordicsemi/dfu/dfu_transport_serial.py", line 366, in __get_mtu
    self.mtu = struct.unpack('<H', bytearray(response))[0]
TypeError: cannot convert 'NoneType' object to bytearray
```

I can run the Python2 scripts (after fixing some stupid tabs vs. spaces issues), but I don't see any advertisements. 
```
# python2 link_layer_length_overflow.py /dev/ttyUSB0 C3:38:99:XX:XX:XX
Serial port: /dev/ttyUSB0
Advertiser Address: C3:38:99:XX:XX:XX
TX ---> BTLE_ADV / BTLE_SCAN_REQ
Waiting advertisements from c3:38:99:XX:XX:XX
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
[...snip...]
```

I'm new to Bluetooth hacking. Do you think your code will ever work with the Adafruit or is it easier to just buy the correct dongle?

Thanks, Klaus

Matheus-Garbelini, 2020-03-11:
Hi @keisentraut sorry about the tab issues. I didn't have this problem as it works normally on my system. I've just pushed a commit to fix the tabs though.

Unfortunately, the nRF52832 does not support native USB. You were able to run the script because the board has an external serial USB converter, giving you a false impression that the MCU was doing something. We'll add support for nRF52832 on our TODO list. Meanwhile, you would need to buy the [nRF52840 dongle](https://www.nordicsemi.com/Software-and-tools/Development-Kits/nRF52840-Dongle) or [Adafruit Feather nRF52840 Express](https://www.adafruit.com/product/4062).

keisentraut, 2020-03-12:
> You were able to run the script because the board has an external serial USB converter, giving you a false impression that the MCU was doing something.

That's what I've already been expecting, but I wasn't sure. Thanks for confirmation!

closed 2020-03-12 by keisentraut