Matheus-Garbelini/sweyntooth_bluetooth_low_energy_attacks #5: Error running Telink_key_size_overflow.py

jandyman, 2020-03-11:
I've managed to successfully DFU the Nordic dongle and retreive the MAC address of my DUT. I've created a python 2.7 environment to run in. However, when I try the first test, I get an internal error. Can you tell me what it means? I'm running this on Mac OS Catalina, but I don't think that has anything to do with this error. But maybe I'm wrong. Here's the error

(Base27) andy@Andrews-MacBook-Pro-2 sweyntooth_bluetooth_low_energy_attacks-master % python Telink_key_size_overflow.py /dev/tty.usbmodem143301 C1:0B:D7:A9:6B:81
Traceback (most recent call last):
  File "Telink_key_size_overflow.py", line 12, in <module>
    from drivers.NRF52_dongle import NRF52Dongle
  File "/Users/andy/Downloads/sweyntooth_bluetooth_low_energy_attacks-master/drivers/NRF52_dongle.py", line 8, in <module>
    from scapy.layers.bluetooth4LE import BTLE, NORDIC_BLE
  File "/Users/andy/Downloads/sweyntooth_bluetooth_low_energy_attacks-master/libs/scapy/layers/bluetooth4LE.py", line 21, in <module>
    from scapy.layers.bluetooth import EIR_Hdr, L2CAP_Hdr
  File "/Users/andy/Downloads/sweyntooth_bluetooth_low_energy_attacks-master/libs/scapy/layers/bluetooth.py", line 28, in <module>
    from scapy.sendrecv import sndrcv
  File "/Users/andy/Downloads/sweyntooth_bluetooth_low_energy_attacks-master/libs/scapy/sendrecv.py", line 36, in <module>
    import scapy.route  # noqa: F401
  File "/Users/andy/Downloads/sweyntooth_bluetooth_low_energy_attacks-master/libs/scapy/route.py", line 194, in <module>
    conf.route = Route()
  File "/Users/andy/Downloads/sweyntooth_bluetooth_low_energy_attacks-master/libs/scapy/route.py", line 27, in __init__
    self.resync()
  File "/Users/andy/Downloads/sweyntooth_bluetooth_low_energy_attacks-master/libs/scapy/route.py", line 35, in resync
    self.routes = read_routes()
  File "/Users/andy/Downloads/sweyntooth_bluetooth_low_energy_attacks-master/libs/scapy/arch/unix.py", line 82, in read_routes
    netif = rt[4 + mtu_present + prio_present + refs_present + locked]
IndexError: list index out of range

Matheus-Garbelini, 2020-03-11:
Hi @jandyman, by looking at the end of the error log, it seems there's something wrong within read_routes. This function is internal to Scapy library and is not directly related to SweynTooth. As the Scapy version used in this repo is older, it may have some incompatibility with your system. 
I suggest you to perform a quick test by downloading the latest [Scapy library](https://github.com/secdev/scapy) and installing it on `libs/scapy` folder to check if this particular error goes away. If the latest scapy fixes the issue for your system, you can then patch the following files in the new version:
* [/libs/scapy/layers/bluetooth.py](https://github.com/Matheus-Garbelini/sweyntooth_bluetooth_low_energy_attacks/blob/master/libs/scapy/layers/bluetooth.py)
* [/libs/scapy/layers/bluetooth4LE.py](https://github.com/Matheus-Garbelini/sweyntooth_bluetooth_low_energy_attacks/blob/master/libs/scapy/layers/bluetooth4LE.py)

I don't currently have access to a Mac system. In the meantime, let me know if Scapy still gives you an error.

jandyman, 2020-03-11:
OK, I'll do that. But one question. I would think that the latest scapy libs would be for Python 3.7, and your scripts are designed for 2.7? Any issue there?

Matheus-Garbelini, 2020-03-11:
Hi @jandyman . I'm not aware of such an issue. The latest repo still states to support for Python 2.7. Nevertheless, I just downloaded pushed the latest version of Scapy on this repo and applied the patches. It's working fine on both Windows and Ubuntu 18.04. Let me know if your system still gives the same error.

jandyman, 2020-03-11:
Thank you for updating the repo. That appears to fix the issue. The only remaining question I've got is why I'm not seeing a connect request, comparing to your output. It may be because our device is not a standard device and has a slightly unusual connect sequence?

Here's the output:

Serial port: /dev/tty.usbmodem144201
Advertiser Address: C1:0B:D7:A9:6B:81
TX ---> BTLE_ADV / BTLE_SCAN_REQ
Waiting advertisements from c1:0b:d7:a9:6b:81
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
C1:0B:D7:A9:6B:81: BTLE_ADV / BTLE_SCAN_RSP Detected
TX ---> BTLE_ADV / BTLE_CONNECT_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
Slave Connected (L2Cap channel established)
TX ---> BTLE_DATA / CtrlPDU / LL_VERSION_IND
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
C1:0B:D7:A9:6B:81: BTLE_ADV / BTLE_SCAN_RSP Detected
TX ---> BTLE_ADV / BTLE_CONNECT_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
Slave Connected (L2Cap channel established)
TX ---> BTLE_DATA / CtrlPDU / LL_VERSION_IND
TX ---> BTLE_ADV / BTLE_SCAN_REQ
C1:0B:D7:A9:6B:81: BTLE_ADV / BTLE_SCAN_RSP Detected
TX ---> BTLE_ADV / BTLE_CONNECT_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ

Matheus-Garbelini, 2020-03-15:
@jandyman can you try to increse the connection request parameters?
by default, the value of the connection interval is 16 (20ms). Your peripheral may not be accepting low interval values. You can try to increase these parameters on the conn_request variable:
https://github.com/Matheus-Garbelini/sweyntooth_bluetooth_low_energy_attacks/blob/master/Telink_key_size_overflow.py#L110

jandyman, 2020-03-16:
Actually, it's the opposite. Our minimum connection interval is 7.5ms, the max is 15ms. I'll try with a different connection interval

Matheus-Garbelini, 2020-03-16:
@jandyman thanks for informing, this may be impeding the script to perform the connection. Let me know if this works as we are planning to update the repo. to describe the connection parameters. These are of critical importance to initiate a successful connection with the peripheral (i.e, receive an empty PDU)

jandyman, 2020-03-16:
I did get better results on the one test by changing the parameters, I need to rerun some others. It could be useful to have any such parameters be keyword arguments.

Matheus-Garbelini, 2020-03-17:
@jandyman good to know.
Also, note that the repository was updated with a better llid_deadlock.py script.

Matheus-Garbelini, 2020-03-20:
Hi @jandyman, did you successfully run Telink_key_size_overflow.py after changing the connection interval or experienced some other issues?

jandyman, 2020-03-20:
I was able make *all* the standard scripts work by modifying calls to BTLE_ADV(). There were generally two calls per script that I had to modify. Of course it would be good to have this all unified and work for both types of addressing, as you indicated in your email.

Matheus-Garbelini, 2020-03-20:
@jandyman Thanks a lot, I will update you once this inclusion is added to the repo.

closed 2020-03-20 by Matheus-Garbelini