Matheus-Garbelini/sweyntooth_bluetooth_low_energy_attacks #6: nRF52840 dongle is not detecting cubiTag to execute public key crash

Priyasachi, 2020-03-15:
I have created necessary environment on windows 10 to crash the cubiTag device by using the nRF52840 dongle, unfortunately i always received the following output. 
 
[image]

can you please suggest something by looking out the screenshot.

Matheus-Garbelini, 2020-03-15:
Hi @Priyasachi, did you flash the **nRF52840 driver firmware** before executing the exploit script? The command to install the firmware on the dongle is the following:

```Shell
python -m pip install nrfutil pyserial pycryptodome
nrfutil dfu usb-serial -p COM4 -pkg nRF52_driver_firmware.zip
```

Priyasachi, 2020-03-16:
Hi, Yes i did flash the nRF52840 driver firmware before executing the exploit script and both run successfully. (My serial port  changed) 

                  python -m pip install nrfutil pyserial pycryptodome
                  nrfutil dfu usb-serial -p COM7 -pkg nRF52_driver_firmware.zip

[image]

Matheus-Garbelini, 2020-03-16:
@Priyasachi does this happen only with the public key script, or if you run the other scripts it can find CubiTag? Also make sure the CubiTag is disconnected from any smartphone/app. SweynTooth scripts can't find CubiTag if a connection has already been established between it and any other device. For example, if you can locate CubiTag address on the nRF connect app, without clicking on the connect button, the public key script should also be able to find it. 

I'll try to replicate this problem here, but let me know if you have verified the CubiTag address is correct and that you are not connected to it while executing the script.

Regards

Priyasachi, 2020-03-17:
Thank you very much for your comments.
1. Yes For other scripts also similar output was observed
2. After firmware is flashed the red light stop blinking (does it mean dongle is no more in DFU mode?) and we are again pressing reset button to make it on DFU mode.
3. Also CubiTag address is correct and it was not connected to any other device 
and then we are executing script.

After firmware is flashed the red light stop blinking, and try to execute script without pressing reset button.
[image]

Matheus-Garbelini, 2020-03-17:
Hi @Priyasachi after flashing the red led stops working, but the green led starts blinking.
By looking at your output, the firmware was flashed, but the COM port may have changed after the nRF switched from DFU mode to normal operation (COM7 windows error). 

Please, check if the script works by using the COM port under normal operation of the nRF52 dongle and let me know.

Priyasachi, 2020-03-17:
Yes, We already checked that cubiTag can be connected to nRF52840 dongle. It would be wondering if you can give us the hints on final step of crashing cubiTag

[image]

Priyasachi, 2020-03-17:
We are getting the expected output. Thank you.

Matheus-Garbelini, 2020-03-17:
Hi @Priyasachi I noticed you were not getting a yellow message showing that a malformed packet was sent on the terminal.
Could you share what you did to get the expected output so we can try to improve the scripts?

Thanks.

Priyasachi, 2020-03-18:
Hi @Matheus-Garbelini after reconnecting nRF52840 dongle we are not running the script to flash the firmware again that might be the reason for not getting the yellow message (malfunction packet was sent on the terminal). also we believe that the environment (many active Bluetooth/WiFi) also affects the final output. Thank you 
[image]

Matheus-Garbelini, 2020-03-19:
@Priyasachi thanks a lot, I'll investigate what you reported about the environment.
Regards.

closed 2020-03-19 by Matheus-Garbelini