Matheus-Garbelini/sweyntooth_bluetooth_low_energy_attacks #8: esp32_hci_nimble_deadlock: Test Script error

WCCWCC, 2020-03-18:
/media/test1/6EAA-ABB1/Screenshot from 2020-03-18 15-47-47.png

Please help me solve it.

WCCWCC, 2020-03-18:
Traceback (most recent call last):
  File "esp32_hci_nimble_deadlock.py", line 178, in <module>
    pkt = receive_encrypted(pkt)  # Decrypt Link Layer
  File "esp32_hci_nimble_deadlock.py", line 155, in receive_encrypted
    print(Fore.RED + "MIC Wrong: " + e)
TypeError: cannot concatenate 'str' and 'exceptions.ValueError' objects

Matheus-Garbelini, 2020-03-18:
@WCCWCC be careful, this script is not known to the public, contact me via my email: mgarbelix@gmail.com

You can fix this error by changing the line to
print(Fore.RED + "MIC Wrong: " + str(e))
Notice the addition of the str casting over the "e" error.

closed 2020-03-18 by Matheus-Garbelini

WCCWCC, 2020-03-18:
@Matheus-Garbelini 
    Thanks, But i have some other questions, i have sent an email, hope to help me.