Matheus-Garbelini/sweyntooth_bluetooth_low_energy_attacks #12: nRF52840 Development Kit not working with the python scripts

charanteja070, 2020-06-17:
Hi Matheus,

I have flashed driver.hex and softdevice.hex using nRFConnect App into nRF52840 Development kit. When I am running the any script in fedora OS, I am seeing the below issue
[root@cpu295 sweyntooth_bluetooth_low_energy_attacks-master]# python Telink_key_size_overflow.py /dev/ttyACM0 88:DA:1A:B6:82:DA
Serial port: /dev/ttyACM0
Advertiser Address: 88:DA:1A:B6:82:DA
TX ---> BTLE_ADV / BTLE_SCAN_REQ
Waiting advertisements from 88:da:1a:b6:82:da
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
^CTraceback (most recent call last):
  File "Telink_key_size_overflow.py", line 80, in <module>
    data = driver.raw_receive()
  File "/work/nlink_host/WSDK_2.6.0/Sandeep/sweyntooth_bluetooth_low_energy_attacks-master/drivers/NRF52_dongle.py", line 75, in raw_receive
    c = self.serial.read(1)
  File "/usr/lib/python2.7/site-packages/serial/serialposix.py", line 483, in read
    ready, _, _ = select.select([self.fd, self.pipe_abort_read_r], [], [], timeout.time_left())
KeyboardInterrupt

Can you please help me to resolve this issue.

Thanks
Charan

Matheus-Garbelini, 2020-06-17:
Hi @charanteja070 the SweynTooth firmware does not work out of the box with nRF52840 Development kit. 

I was able to compile for this target platform but I'm not sure if it works as I don't have this board.
Nevertheless, feel free to try the attached new firmware and let me know if it works (nrf52840_dk.zip).

[nrf52840_dk.zip](https://github.com/Matheus-Garbelini/sweyntooth_bluetooth_low_energy_attacks/files/4792861/nrf52840_dk.zip)

You can flash it the same way you did before (both the firmware and the softdevice.hex).

Matheus-Garbelini, 2020-06-17:
Also, make sure to use the nRF52 integrated USB, not the Interface MCU USB (The one that flashes the nRF52 MCU).

The correct USB port to use when running the scripts should be the bottom one near nRF52 as shown below.
[image]

charanteja070, 2020-06-17:
Hi Matheus,
Can you please let me know which files I need flash
nRF52_driver_firmware.hex + S140_softdevice.hex + nRF52840_dk.hex
                                      (OR)
S140_softdevice.hex + nRF52840_dk.hex
                                      (OR)
only nRF52840_dk.hex

Thanks
Charan

charanteja070, 2020-06-17:
Can you please let me know the driver for the nRF52 USB on which I have to run python scripts, since the LED5 is not ON.

charanteja070, 2020-06-17:
Observation:
When I am using 3 files "nRF52_driver_firmware.hex + S140_softdevice.hex + nRF52840_dk.hex",
and connected to Fedora 32 bit Machine with nRF USB, no LEDS are turned ON and I am facing the same issue mentioned in comment 1 after running python script.

Matheus-Garbelini, 2020-06-17:
Hi @charanteja070, you must flash only S140_softdevice.hex + nRF52840_dk.hex
And the less will not work since the firmware toggles GPIOs which are not physically connected to your board.

There's no driver for nRF52840 USB. The native USB peripheral of the nRF52 MCU uses USB Serial CDC Class which is already integrated on your Linux Kernel. You can check if the correct USB was listed by your OS via the kernel log (run *dmesg*). You should see the following message from the kernel, which indicates the serial port you should use for this firmware:

[image]

Alternatively, run *lsusb* and check if the output lists a USB device with *ID 239a:8029*.
[image]

Can you confirm if you can see the above outputs with your board?

charanteja070, 2020-06-17:
Hi Matheus,
I am getting the same image what you have shown after giving lsusb command.
I will try to flash **S140_softdevice.hex + nRF52840_dk.hex** files and will update the results.

Matheus-Garbelini, 2020-06-17:
Thanks @charanteja070. So this confirms that the nRF52 firmware is correctly running, but now the issue is why the script is not detecting your target BLE peripheral.

It's possible that you are not receiving any BLE advertisements at all.
Can you add **after** line 83 (https://github.com/Matheus-Garbelini/sweyntooth_bluetooth_low_energy_attacks/blob/master/Telink_key_size_overflow.py#L83) the following python line: **print(pkt.summary() + ', Address: ' + pkt.AdvA)**

This helps you to check if you are at least receiving any BLE advertisements as below: 

[image]

charanteja070, 2020-06-18:
Hi Matheus,
I will do that change and let you know, but before that please tell me from which iterface I need to flash either J3(nRFUSB interface) or J2(Segger Jlink Interface).
Normally I will use J2 instead J3, Correct me If I am flashing in wrong way.

Matheus-Garbelini, 2020-06-18:
Hi @charanteja070 you flash via J2 and run the scripts via J3.

Were you running the scripts via J2?

LucaBT, 2021-03-03:
Hi @Matheus-Garbelini,
I'm facing the same issue. I'm trying to use nRF52840 Development Kit with the custom FW nrf52840_dk.zip. 
I've followed these steps
1. On nRFConnect Programmer I've added the s140_nrf52_6.1.1_softdevice.hex to the layout
2. I've added the nrf52840_dk.hex to the layout and flashed the device connected by the Interface MCU USB connector
3. Connected the device through the nRF52 integrated USB
4. Launched the script DA14580_exploit_att_crash.py with the proper COM and MAC address

The device seems to be correctly recognized
[image]

But when I launch the script the following error occurs:
[image]

Is there something wrong with the procedure?
Thanks for your support.

Matheus-Garbelini, 2021-03-03:
Hi @LucaBT, it could be some python version issue.
What python version are you using?

LucaBT, 2021-03-03:
I'm using `Python 2.7.3 ` 
with the following libraries:
Package                       Version
----------------------------- -------
aardvark-py                   5.40
backports.functools-lru-cache 1.6.1
behave                        1.0.0
click                         7.1.2
colorama                      0.4.3
crcmod                        1.7
cycler                        0.10.0
ecdsa                         0.13
enum34                        1.1.10
future                        0.18.2
intelhex                      2.2
ipaddress                     1.0.21
kiwisolver                    1.1.0
libusb1                       1.7
linecache2                    1.0.0
matplotlib                    2.2.5
nrfutil                       5.2.0
numpy                         1.16.6
parse                         1.3.3
pc-ble-driver-py              0.11.4
piccata                       1.0.0
pip                           20.0.2
protobuf                      3.15.3
psutil                        5.7.2
pycryptodome                  3.10.1
pyparsing                     2.4.7
pyserial                      3.4
pyspinel                      1.0.0a3
python-dateutil               2.8.1
pytz                          2020.1
pywin32                       228
PyYAML                        4.2b1
saleae                        0.9.1
setuptools                    44.1.1
six                           1.15.0
tqdm                          4.25.0
traceback2                    1.4.0
unittest2                     1.0.1
vboxapi                       1.0
wrapt                         1.10.8

Matheus-Garbelini, 2021-03-04:
hi @LucaBT try using Python3 instead.

LucaBT, 2021-03-04:
Hi @Matheus-Garbelini, thanks for your support.
I've created a `venv` with `Python 3.7.9` and the following packages:
Package      Version
------------ -------
pip          19.0.3
pycryptodome 3.10.1
pyserial     3.5
setuptools   40.8.0
six          1.15.0

I've also updated the lines of `DA14580_exploit_att_crash.py` [L139,L140](https://github.com/Matheus-Garbelini/sweyntooth_bluetooth_low_energy_attacks/blob/master/DA14580_exploit_att_crash.py#L139) replacing `'str'.decode('hex')` with `binascii.hexlify(b'str')`. This is the script that I'm interested to run and the problem that I wish to generate.
Now the SDK is correctly recognized and several functions seem to work, although I'm still unable to generate the system crash with the sequence.

I appreciate your willingness. 😄

Matheus-Garbelini, 2021-03-04:
thanks for the reply @LucaBT . Are you able to receive responses from the target such as advertisements (any RX message)?
Keep in mind that I did not test with nrf52840 DK yet, so I'm not sure if you'd face any specific issue because of this board.
If you receive RX messages from the script, then the scripts are working, although if no crash is triggered, you may be using a patched SDK already.

LucaBT, 2021-03-04:
This is my console output.

```
Serial port: COM18
Advertiser Address: XX:XX:XX:XX:XX:XX
TX ---> BTLE_ADV / BTLE_SCAN_REQ
Waiting advertisements from XX:XX:XX:XX:XX:XX
TX ---> BTLE_ADV / BTLE_SCAN_REQ
10:98:C3:69:C6:34: BTLE_ADV / BTLE_ADV_IND Detected
TX ---> BTLE_ADV / BTLE_CONNECT_REQ
10:98:C3:69:C6:34: BTLE_ADV / BTLE_SCAN_RSP Detected
TX ---> BTLE_ADV / BTLE_CONNECT_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
Slave Connected (L2Cap channel established)
TX ---> BTLE_DATA / CtrlPDU / Raw
Connection reset, malformed packet was sent
Waiting advertisements from XX:XX:XX:XX:XX:XX
TX ---> BTLE_ADV / BTLE_SCAN_REQ
Slave Connected (L2Cap channel established)
TX ---> BTLE_DATA / CtrlPDU / Raw
Connection reset, malformed packet was sent
Waiting advertisements from XX:XX:XX:XX:XX:XX
TX ---> BTLE_ADV / BTLE_SCAN_REQ
Slave Connected (L2Cap channel established)
TX ---> BTLE_DATA / CtrlPDU / Raw
Connection reset, malformed packet was sent
Waiting advertisements from XX:XX:XX:XX:XX:XX
TX ---> BTLE_ADV / BTLE_SCAN_REQ
Slave Connected (L2Cap channel established)
TX ---> BTLE_DATA / CtrlPDU / Raw
Connection reset, malformed packet was sent
Waiting advertisements from XX:XX:XX:XX:XX:XX
TX ---> BTLE_ADV / BTLE_SCAN_REQ
Slave Connected (L2Cap channel established)
TX ---> BTLE_DATA / CtrlPDU / Raw
Connection reset, malformed packet was sent
Waiting advertisements from XX:XX:XX:XX:XX:XX
TX ---> BTLE_ADV / BTLE_SCAN_REQ
Slave Connected (L2Cap channel established)
TX ---> BTLE_DATA / CtrlPDU / Raw
Connection reset, malformed packet was sent
Waiting advertisements from XX:XX:XX:XX:XX:XX
TX ---> BTLE_ADV / BTLE_SCAN_REQ
Slave Connected (L2Cap channel established)
TX ---> BTLE_DATA / CtrlPDU / Raw
Connection reset, malformed packet was sent
Waiting advertisements from XX:XX:XX:XX:XX:XX
TX ---> BTLE_ADV / BTLE_SCAN_REQ
Slave Connected (L2Cap channel established)
TX ---> BTLE_DATA / CtrlPDU / Raw
Connection reset, malformed packet was sent
Waiting advertisements from XX:XX:XX:XX:XX:XX
TX ---> BTLE_ADV / BTLE_SCAN_REQ
Slave Connected (L2Cap channel established)
TX ---> BTLE_DATA / CtrlPDU / Raw
Connection reset, malformed packet was sent
Waiting advertisements from XX:XX:XX:XX:XX:XX
TX ---> BTLE_ADV / BTLE_SCAN_REQ
Slave Connected (L2Cap channel established)
TX ---> BTLE_DATA / CtrlPDU / Raw
Connection reset, malformed packet was sent
Waiting advertisements from XX:XX:XX:XX:XX:XX
TX ---> BTLE_ADV / BTLE_SCAN_REQ
Slave Connected (L2Cap channel established)
TX ---> BTLE_DATA / CtrlPDU / Raw
Slave RX <--- BTLE_ADV / BTLE_ADV_NONCONN_IND
Connection reset, malformed packet was sent
Waiting advertisements from XX:XX:XX:XX:XX:XX
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
```

Matheus-Garbelini, 2021-03-04:
@LucaBT thanks.
Good to know that the firmware works for  nRF52840 DK.
About the logs, the device you are testing is could be already patched or the device silently restarts which may lead to a conclusion that is not vulnerable. 

If your product has the BLE module isolated from the main CPU, it's unlikely you will see any real crash occurring as the main CPU may be just restarting the HCI communication with the BLE module.

YY-chen555, 2023-02-03:
> 

> Thanks @charanteja070. So this confirms that the nRF52 firmware is correctly running, but now the issue is why the script is not detecting your target BLE peripheral.
>
> It's possible that you are not [...]

Matheus-Garbelini, 2023-02-03:
Hi @YY-chen555 what USB port are you connecting to your  nRF52840_dk.?
The correct USB port is the one close to the buttons.

YY-chen555, 2023-02-06:
> Hi @YY-chen555 what USB port are you connecting to your nRF52840_dk.?
>The correct USB port is the one close to the buttons.
Hello
This is my devices.
[image]
And it only has one port

Matheus-Garbelini, 2023-02-06:
Hi @YY-chen555 it seems your dongle is not nRF52840 DK. DK stands for development kit and is a bug board.
Yours seems to be the dongle. So your firmware is the one mentioned on the readme section: https://github.com/Matheus-Garbelini/sweyntooth_bluetooth_low_energy_attacks/blob/master/nRF52_driver_firmware.hex

YY-chen555, 2023-02-06:
> Hi @YY-chen555 what USB port are you connecting to your nRF52840_dk.? The correct USB port is the one close to the buttons.

Hello
This is my devices.
[image]

And it only has one port. I can use this to scan bluetooth by using it's own firmware.

YY-chen555, 2023-02-06:
> 

Thank you for replying.
In fact,I have flashed the nRF52_driver_firmware.hex and s140_nrf52_6.1.1_softdevice.hex.
Maybe I should only flash nRF52_driver_firmware.hex?