Matheus-Garbelini/sweyntooth_bluetooth_low_energy_attacks #13: Firmware question

AdySan, 2020-06-27:
Hello, 

Sorry if I missed it, but is the firmware open source? If not can you at least give some hints what it's based on?

Thanks,
Aditya

Matheus-Garbelini, 2020-06-30:
Hi @AdySan, sorry for the delay.

The firmware is closed source for now due to embargo, but it will be available open-source through request soon. Note that you can currently use the firmware to send or receive LL packets by just using the driver library included on this repo.

AdySan, 2020-07-01:
Thanks, I understand @Matheus-Garbelini, also thank you for the great work.

Matheus-Garbelini, 2020-07-02:
Hi @AdySan will reopen this issue as to notify others that the firmware code will be released by the end of this month. You'll be notified by then, and we will include documentation of the current API.

JakePeralta0, 2020-09-08:
Hi @Matheus-Garbelini any updates on the firmware release? Is there any way to reset the the firmware via UART?

byte-arts, 2021-02-04:
Hello @Matheus-Garbelini,
i am also very interested in the source code of the firmware, is there any update yet?

Or maybe you can answer me some questions.

Is it possible to advertise with this firmware and therefore create a simple peripheral? I tried to send advertisements, but had no success until now.

Another question is, can i use any kind of address (random and public) with this firmware just by setting the corresponding packet values (seems like)? Or is there any additional logic in the firmware?

Regards,
byte-arts

Matheus-Garbelini, 2021-02-04:
Hi @byte-arts, @AdySan 

- The firmware cannot advertise as it behaves like a central, not a peripheral. However, if you want to take a look at the source code, please send us an email to sweyntooth@gmail.com from your company/university email.
- Yes, you can use any address, just modify the master_address variable present on the script ([example](https://github.com/Matheus-Garbelini/sweyntooth_bluetooth_low_energy_attacks/blob/master/sequential_att_deadlock.py#L18)). You can modify any field within the Scapy packet, the firmware will send the exact way you wrote the packet, **except** for the acknoledgments bits on the BLE header which are handled by the firmware automatically so you can keep a connectionion to the device.

Please, do not hesitate to open a new issue if you have any other question, regards.

fr0zenrain, 2021-07-07:
now 2021 ,open source delay?

Matheus-Garbelini, 2021-07-07:
Hi @fr0zenrain 
Please send us an email to sweyntooth@gmail.com from your university email.