Matheus-Garbelini/sweyntooth_bluetooth_low_energy_attacks #14: Silicon Labs SoC's

rafaribe, 2020-07-03:
There is nothing in the report about Silicon Labs SoC's but have they been subject to testing from your side? 
I'm specially interested in the [Blue Gecko SoC's](https://docs.silabs.com/bluetooth/latest/general/overview/#about-the-bluetooth-stack)
As far as I can see they implemented some protection on [another product](https://www.silabs.com/documents/public/release-notes/Bluetooth-Smart-Software-Release-Notes.pdf)

Cheers and keep up the good work!

Matheus-Garbelini, 2020-07-03:
Hi @rafaribe thanks for reporting,
We didn't test Blue Gecko, but we were contacted some time ago by Silabs to assist in their use of the test scripts.

I guess they silently updated their SDK.
We will buy this board and run some tests with it.

If you happen to have this board, feel free to report if any of the existing scripts are successful.

rafaribe, 2020-07-06:
I have the board, unfortunately I don't have the Nordic nRF52840 Dongle. I have a couple of BlueGiga dongles and 3x Micro:bit that I've used with [bltejack](https://github.com/virtualabs/btlejack), any chance I can use this hardware in conjunction with the provided scripts?
Otherwise, I'll try to buy the Nordic dongle.