Matheus-Garbelini/sweyntooth_bluetooth_low_energy_attacks #17: device cannot be correctly detected again after the firmware is flashed

f0rm2l1n, 2020-11-16:
Hello there, 
I flashed the firmware as the guideline. 
```
nrfutil dfu usb-serial -p /dev/ttyACM0 -pkg nRF52_driver_firmware.zip
```

It succeeds for the first time and leaves me a 100% progress bar.

However, the next time I plug my nrf52840, everything gets wrong...

The nrfconnect says like below
[image]

In addition, the script succeeded before now comes with below errors
```
h# nrfutil dfu usb-serial -p /dev/ttyACM0 -pkg ./sweyntooth_bluetooth_low_energy_attacks/nRF52_driver_firmware.zip
2020-11-16 10:34:41,980 No trigger interface found for device with serial number: D5DADB9ADEDF5D06, Product ID: 0x8029 and Vendor ID: 0x239A

Traceback (most recent call last):
  File "/home/lin/workplace/bluetooth/2.7env/bin/nrfutil", line 8, in <module>
    sys.exit(cli())
  File "/home/lin/workplace/bluetooth/2.7env/local/lib/python2.7/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/lin/workplace/bluetooth/2.7env/local/lib/python2.7/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/lin/workplace/bluetooth/2.7env/local/lib/python2.7/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/lin/workplace/bluetooth/2.7env/local/lib/python2.7/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/lin/workplace/bluetooth/2.7env/local/lib/python2.7/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/lin/workplace/bluetooth/2.7env/local/lib/python2.7/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/lin/workplace/bluetooth/2.7env/local/lib/python2.7/site-packages/nordicsemi/__main__.py", line 1001, in usb_serial
    timeout)
  File "/home/lin/workplace/bluetooth/2.7env/local/lib/python2.7/site-packages/nordicsemi/__main__.py", line 956, in do_serial
    dfu.dfu_send_images()
  File "/home/lin/workplace/bluetooth/2.7env/local/lib/python2.7/site-packages/nordicsemi/dfu/dfu.py", line 121, in dfu_send_images
    self._dfu_send_image(self.manifest.softdevice)
  File "/home/lin/workplace/bluetooth/2.7env/local/lib/python2.7/site-packages/nordicsemi/dfu/dfu.py", line 90, in _dfu_send_image
    self.dfu_transport.open()
  File "/home/lin/workplace/bluetooth/2.7env/local/lib/python2.7/site-packages/nordicsemi/dfu/dfu_transport_serial.py", line 216, in open
    self.__set_prn()
  File "/home/lin/workplace/bluetooth/2.7env/local/lib/python2.7/site-packages/nordicsemi/dfu/dfu_transport_serial.py", line 360, in __set_prn
    self.__get_response(DfuTransportSerial.OP_CODE['SetPRN'])
  File "/home/lin/workplace/bluetooth/2.7env/local/lib/python2.7/site-packages/nordicsemi/dfu/dfu_transport_serial.py", line 488, in __get_response
    raise NordicSemiException('No Response: 0x{:02X}'.format(resp[0]))
pc_ble_driver_py.exceptions.NordicSemiException: No Response: 0x44
```

I use the ubuntu18.04, with the latest JLINK and NRF Connect for desktop

> The same error also occurs in my windows10 environment

I don't know how to do next to recover my dongle or fix this? Can you offer any advice?
Looking forward to any possible way out,

Matheus-Garbelini, 2020-11-16:
Hi @f0rm2l1n if you put the dongle in programming mode (reset the dongle while it is plugged to USB), do you get to see the red LED blinking? You can only flash a new firmware if the dongle is already on programming mode (DFU mode). Let me know if this helps.

f0rm2l1n, 2020-11-16:
@Matheus-Garbelini Thanks for your reply!
Oops, this may sound quite weird as the dongle of mine never blink red (only yellow for now)
Every time I plug it into my laptop the yellow LED starts blinking and no matter how I press the button, nothing changes.

some pictures are shown below
[image]
[image]

Should I buy other cables for flushing the firmware? Or maybe I just bought a bad one which is not support programming mode?
P.S. I try many different methods of pressing this damn button :(. No idea which one is correct...

Matheus-Garbelini, 2020-11-16:
Hi, @f0rm2l1n it is not common that you cannot put the dongle on programming mode again, as the DFU is locked by Nordic and cannot be removed even if flashing SweynTooth firmware.
Can you confirm what happens when you press the small reset button **while the dongle is plugged in?**
Please find below such button which is supposed to put the dongle in programming mode again with a blinking red led:
[image]

f0rm2l1n, 2020-11-16:
@Matheus-Garbelini 
I am such an idiot T.T I just keep pressing the SW button
Thanks for your answer, it really helps a green hand out~~

closed 2020-11-16 by f0rm2l1n

Matheus-Garbelini, 2020-11-16:
@f0rm2l1n no worries, I've just updated the documentation so this is better clarified: https://github.com/Matheus-Garbelini/sweyntooth_bluetooth_low_energy_attacks#getting-started-installation

f0rm2l1n, 2020-11-16:
@Matheus-Garbelini 
In addition, one more question, If I choose to flash the firmware though "connector", which options below I should choose
[image]

> The command line just drives me crazy sometimes :)

Matheus-Garbelini, 2020-11-16:
for s140_nrf52_6.1.1_softdevice.hex you should write a custom SoftDevice ID: **0xB6**