Matheus-Garbelini/sweyntooth_bluetooth_low_energy_attacks #18: Vulnerable peripherals

cxjcxggg, 2021-02-01:
Hi @Matheus-Garbelini, thanks to your great work! 
I am new to BT CVE. How can I know which type of board or BT stack or constained device is vulnerable to a certain CVE?
Could you share the vulnerable peripherals you used to test this project as example? 
Appreciate it !

Matheus-Garbelini, 2021-02-02:
Hi @cxjcxggg sure. We have all the details on our **disclosure website**: [https://asset-group.github.io/disclosures/sweyntooth/#commento-login-box-container](https://asset-group.github.io/disclosures/sweyntooth/#commento-login-box-container)

On the patches section, it contains all the SDK updates from each vendor. If you are looking however for the exact development board we have used to perform the tests, check out **Table 2** on SweynTooth paper: [https://www.usenix.org/system/files/atc20-garbelini.pdf](https://www.usenix.org/system/files/atc20-garbelini.pdf)

cxjcxggg, 2021-02-02:
Wow, it is of great help to me! 

Thanks, @Matheus-Garbelini .

Matheus-Garbelini, 2021-02-04:
@cxjcxggg good to know. Please feel free to reopen an issue if you have any other questions. regards.

closed 2021-02-04 by Matheus-Garbelini