Matheus-Garbelini/sweyntooth_bluetooth_low_energy_attacks #19: Question about the vulnerabilities.

andrewpotterdr, 2021-02-15:
Hello,

Sorry if I'm missing something in the documentation but are there 18 or 14 vulnerabilities? If there are 18, where are the other 4? If the extra folders present another vulnerabilities, then wouldn't it be 26 total? Please help me understand.

Thanks in advance!

Matheus-Garbelini, 2021-02-16:
Hi @andrewpotterdr The total is 17 and all CVEs can be seen on this table: 
[https://github.com/Matheus-Garbelini/sweyntooth_bluetooth_low_energy_attacks#available-ble-exploits](https://github.com/Matheus-Garbelini/sweyntooth_bluetooth_low_energy_attacks#available-ble-exploits)
You noticed there is 18 instead of 17, this was just a hint that the last one is not disclosed. So officially, only 17.

johbec, 2021-12-13:
> 

Oooh exciting! Is there a batch 3 coming?

Matheus-Garbelini, 2021-12-13:
@johbec no haha. That's for other researchers in the BLE community to discover.