Matheus-Garbelini/sweyntooth_bluetooth_low_energy_attacks #20: About the vulnerable source code

cxjcxggg, 2021-03-12:
Hi @Matheus-Garbelini , I want to dig a little bit deeper into CVE-2020-13594. I wonder how I can have the source code of the [esp32 static Bluetooth library](https://github.com/espressif/esp32-bt-lib)? They seem to be close-source. Did you have the source code during your work?

Matheus-Garbelini, 2021-03-12:
Hi @cxjcxggg unfortunately not.
ESP32 uses intellectual Property from Riviera Waves RF Subsystem, so even if Espressif wants they cannot share their source code due to legal reasons.

cxjcxggg, 2021-03-12:
Hi @Matheus-Garbelini , sad to know that. I am also interested in zephyr vulnerabilities you found. The zephyr project is open-source, does that mean I can have the source code which triggers the CVEs?

Matheus-Garbelini, 2021-03-12:
@cxjcxggg yes, for zephyr you can find the specific merges here:
https://docs.zephyrproject.org/latest/security/vulnerabilities.html#cve-2020-10069

cxjcxggg, 2021-03-12:
@Matheus-Garbelini , I got it, thanks a lot!

cxjcxggg, 2021-03-25:
Hi @Matheus-Garbelini , it's me again. I won't open a new issue so that I won't bother you. I am trying to reproduce the invalid channel map vulnerability (CVE-2020-10069) on Zephyr project, it is fixed in zephyr v2.3.0, so I need to use v2.2.0 or v2.2.1. I successfully connected the nrf52840 dongle with the nrf52840 DK via bluetooth using bluetooth peripheral sample in the latest zephyr project. But when I try to build the sample in zephyr v2.2.0, some problems arose because of the different sdk version.
[image]

Therefore, I think I also need to change the Zephyr SDK's version so that it could work with zephyr v2.2.0. But which one should I use? How can I reproduce the attack? Thank you so much.

Matheus-Garbelini, 2021-03-26:
Hi, you can try using west to force update everything:
https://boseji.com/docs/zephyr/tipstricks/update-zephyr-sdk/

Otherwise you need to follow the steps pointed by the older documentation
for 2.2.0 here:
https://docs.zephyrproject.org/2.2.0/getting_started/index.html#install-software-development-toolchain

The SDK version seems to be 0.11.2

Also, make sure to get zephyr version before the patch was introduced. They
did a backport to other versions such as 2.2.0, so checkout a zephyr commit
before sweyntooth fixes was introduced for 2.2.0.

The specific commits that fixed sweyntooth can be seen in their disclosure
page for invalid channel map. You can then checkout the previous commit to
that.

Regards.

cxjcxggg, 2021-03-26:
Hi @Matheus-Garbelini , thanks for your help. I have addressed the issues about the zephyr project. 

Then I set the nrf52840 dongle in DFU mode and use command `nrfutil dfu usb-serial -p COM_PORT -pkg nRF52_driver_firmware.zip` to flash the firmware, "Device programmed" is printed (the blue led goes off). So I replug the dongle (the green led begins to blink) and open the Bluetooth Low Energy app in the nRF Connect for Desktop, but encounter the problem as shown below.

[image]

[image]

So I try to manually flash the firmware with the Programmer app in nRF connect for Desktop, but still fails.

[image]

[image]

[image]

Close the Programmer and open the BLE app, but the device can't be opened.

[image]

[image]

I don't know which step goes wrong. Do I have to use the firmware you provide? I can establish BLE connection with my nrf52840 DK with sniffer hex on nrf52840 dongle. Can I run the `invalid_channel_map.py` with sniffer firmware?

Matheus-Garbelini, 2021-03-26:
Hi @cxjcxggg I'm not sure what you are trying to do. The sniffer and our attacker firmware are different things.

If you see your led blinking green after flashing our firmware than you can run the exploits. Do not try to use it as an sniffer instead.
Let me know how it goes.

Regards.

cxjcxggg, 2021-03-27:
Hi @Matheus-Garbelini , sorry for the confuse. I have finally successfully reproduced the vulnerability! Lol! 

[image]

[image]

I always thought that I needed to connect the nrf52840 dongle to the peripheral before running the script. What I was trying to say yesterday is that I couldn't establish the bluetooth connection with the peripheral after flashing the firmware. So I actually got stuck because of misunderstanding about the way to use the script you provide.

Anyway, I am so excited. Appreciate your kindness a lot! 

Regards.

Matheus-Garbelini, 2021-03-30:
@cxjcxggg good to know. Use it with responsibility :smiley:  :+1:

closed 2021-04-09 by Matheus-Garbelini

yichen115, 2025-07-22:
> Hi [@Matheus-Garbelini](https://github.com/Matheus-Garbelini) , sorry for the confuse. I have finally successfully reproduced the vulnerability! Lol!
>
> [image]
>
> [image]
>
> I always thought that I [...]

Hello, I am researching this vulnerability. Do you have any records or notes of setting up the vulnerability environment? I sent you an email through the contact information in your blog, but I didn't receive a reply. I don't know if you are still using the email.
你好，我正在研究这个漏洞，请问你是否还有漏洞环境搭建的记录或笔记，我通过你博客中的联系方式给你发送了邮箱，但是没有收到回复，不知道邮箱你是否还在用