Matheus-Garbelini/sweyntooth_bluetooth_low_energy_attacks #21: Encryption Error

charanteja070, 2021-06-08:
Hi @Matheus-Garbelini ,

When I am running "python Telink_zero_ltk_installation.py /dev/ttyACM0 <Module Addr>" in Fedora machine and initiating connection to our DUT, I am getting encryption error. Please find the below logs
TX ---> BTLE_ADV / BTLE_CONNECT_REQ
RX <--- BTLE_DATA / CtrlPDU / LL_VERSION_IND
TX ---> BTLE_DATA / CtrlPDU / LL_FEATURE_REQ
RX <--- BTLE_DATA / L2CAP_Hdr / SM_Hdr / SM_Security_Request
Slave requested authentication of 0xd
We are using authentication of 0x9
TX ---> BTLE_ADV / BTLE_SCAN_REQ
88:DA:1A:EB:8D:E6: BTLE_ADV / BTLE_ADV_IND Detected
TX ---> BTLE_ADV / BTLE_CONNECT_REQ
RX <--- BTLE_DATA / CtrlPDU / LL_VERSION_IND
TX ---> BTLE_DATA / CtrlPDU / LL_FEATURE_REQ
RX <--- BTLE_DATA / L2CAP_Hdr / SM_Hdr / SM_Security_Request
Slave requested authentication of 0xd
We are using authentication of 0x9
RX <--- BTLE_DATA / CtrlPDU / LL_FEATURE_RSP
TX ---> BTLE_DATA / CtrlPDU / LL_LENGTH_REQ
RX <--- BTLE_DATA / CtrlPDU / LL_LENGTH_RSP
TX ---> BTLE_DATA / L2CAP_Hdr / SM_Hdr / SM_Pairing_Request
RX <--- BTLE_DATA / L2CAP_Hdr / SM_Hdr / SM_Pairing_Response
TX ---> BTLE_DATA / CtrlPDU / LL_ENC_REQ
RX <--- BTLE_DATA / CtrlPDU / LL_ENC_RSP
Traceback (most recent call last):
  File "Telink_zero_ltk_installation.py", line 338, in <module>
    conn_session_key = bt_crypto_e(conn_ltk[::-1], conn_skd[::-1])
  File "Telink_zero_ltk_installation.py", line 116, in bt_crypto_e
    return aes.encrypt(plaintext)
  File "/usr/lib/python2.7/site-packages/Crypto/Cipher/blockalgo.py", line 244, in encrypt
    return self._cipher.encrypt(plaintext)
**TypeError: argument must be string or read-only buffer, not bytearray**.

My Analysis here is, it is considering encrypt from Fedora library, not from our /lib/smp_server.
Can you please help me out.

Thanks
Charan

Matheus-Garbelini, 2021-06-08:
Hi @charanteja070 what python version are you using?

charanteja070, 2021-06-08:
I am using 2.7

Matheus-Garbelini, 2021-06-08:
Hi @charanteja070 check what input types are going to bt_crypto_e: https://github.com/Matheus-Garbelini/sweyntooth_bluetooth_low_energy_attacks/blob/master/Telink_zero_ltk_installation.py#L323-L325
if you do `type(conn_skd)` it should return a string type `<type 'str'>`.
Perhaps there is something wrong when receiving the `conn_skd` parameter from the peripheral.

charanteja070, 2021-06-09:
My Python version is 2.7.11

Matheus-Garbelini, 2021-06-09:
Hi @charanteja070 I'd have to test this again to check what is going on exactly.
Alternatively, you can use the docker in this repository to ensure that is not some python package broken.

charanteja070, 2021-06-09:
Hi @Matheus-Garbelini ,
Can you please help me out to work on this " you can use the **docker in this repository** to ensure that is not some python package broken."
And also can you please clarify me, "**encrypt**" will be consider from /lib folder (or) /usr/lib/python2.7/.

charanteja070, 2021-06-10:
Hi @Matheus-Garbelini ,
Can you please help me out

charanteja070, 2021-06-14:
HI @Matheus-Garbelini ,
I am able to run this script after installing this "pip install pycryptodome".

So, we can close this Query.

closed 2021-06-14 by charanteja070