Matheus-Garbelini/sweyntooth_bluetooth_low_energy_attacks #24: Question about sending custom BLE packet

yan-xiao-xi, 2021-12-29:
Hi @Matheus-Garbelini , 

First of all, thanks for this work!

I try to send custom BLE packet by modifying the Python script, and sniff packets with Wireshark.

I find that:

1.When sending SCAN_REQ with a valid advertiser address (advertising address of a real peripheral), the LED of nRF52840 flashes blue, and Wireshark can **sometimes** sniff the sent packet.
Codes as follow
[image]
Logs as follow
[image]
Captured packets as follow
[image]

2.When sending SCAN_REQ with an invalid advertiser address (advertising address not used by peripherals), the LED of nRF52840 stays blue, and Wireshark **can not** sniff the sent packet.
Codes as follow
[image]
nRF52840 looks like
[image]

3.When sending custom ADV_NONCONN_IND, the LED of nRF52840 stays blue, and Wireshark **can not** sniff the sent packet.
Codes as follow
[image]
Pcap saved by wrpcap looks like
[image]

And the question is:
1.Why can nRF52840 send SCAN_REQ only with a valid advertiser address, is there any additional logic in the firmware?
2.What does it mean that the LED of nRF52840 stays blue, is that an abnormal state? How can I confirm that the packet is sent to the air successfully?
3.Can I send custom ADV_NONCONN_IND packet (more specifically, custom BLE mesh packet) with your firmware?  If so, would you like to give me a hint about how to do?

Appreciate for your help.

Regards,
yan_xiao_xi

Matheus-Garbelini, 2021-12-29:
Hi @yan-xiao-xi thanks for your great question.

1. Yes, scan requests are only sent by the firmware after it receives a valid advertisement from a peripheral. More precisely it sends a scan request after ~150us of receiving an advertisement from the real peripheral. As you can guess, if you send a invalid advertisement, the firmware will wait forever for an nonexistent peripheral advertisement.
2. It means: "Waiting reception of peripheral advertisement with the address you have informed via scan request PDU". When the blue led goes off, it means that the scan request has been sent to the peripheral.
3. Certainly you can, but you would need to modify the firmware source code in it. If you like low-level firmware programming, feel free to send to sweyntooth@gmail.com a request with your **university email**. Regards.

yan-xiao-xi, 2021-12-29:
Hi @Matheus-Garbelini,

Many thanks to your answer!

yan-xiao-xi, 2021-12-30:
Hi @Matheus-Garbelini,

Sorry to disturb you, have you received the request email?

sacca97, 2024-10-24:
Hi @Matheus-Garbelini  I tried to contact you about the firmware using my academic email but got no reply.