Matheus-Garbelini/sweyntooth_bluetooth_low_energy_attacks #26: some questions about capturing the ble packets

Wi1L-Y, 2022-05-16:
Hi, you have done an outstanding work. 

When I reproducing the job, I run into a bit of trouble. I can only capture the adv packets either using ubertooth or nrf52840, which means I can not get other BLE packets such as "acl data". So I want to know how to get other BLE Interactive network message.

Thank you !

Matheus-Garbelini, 2022-05-16:
Hi @Wi1L-Y

When you run the scripts with nRF840 USB Dongle, are you not able to complete any BLE connection?
Ideally you should get at least Link Layer packets on the console output.
This project does not support Ubertooth, so I'm guessing you are trying to use ubertooth to sniff BLE packets while the other nRF42840 USB is running the scripts?

Wi1L-Y, 2022-05-16:
Yes, I just want to sniff BLE packets(not advertisement packets) using ubertooth or nRF52840? Do you have some method? I'm sorry to ask this question which is not so related to the project

Matheus-Garbelini, 2022-05-16:
Have you tried to follow some BLE Sniffing tutorial such as [the one from Nordic](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fug_sniffer_ble%2FUG%2Fsniffer_ble%2Finstalling_sniffer.html)?

Wi1L-Y, 2022-05-17:
Yes, I have tried this tutorial on nRF52840 and I can only capture the advertisement packets. So I have no idea about capturing the other packets such as "ACL".

Matheus-Garbelini, 2022-05-17:
Perhaps try to select the peripheral address on the wireshark toolbar so the sniffer can follow a connection?

Wi1L-Y, 2022-05-17:
Thank you for your answer !

Yes, I tried to select the target address on the  wireshark toolbar before, and it succeed to capture "adv" packets. However, when I do more behaviors such as transmit data, it can not get the packet. 

Does it occur because of the target ? I have tried many target devices and I cannot get the packet after connected.

Matheus-Garbelini, 2022-05-17:
How are you transmitting data?
Are you connecting to your target via nRF Connect app for android?

Wi1L-Y, 2022-05-17:
Oh!  
Thank you for your guidance. I will try to connect the target via nRF connect while sniffing it by nRF sniifer.

Wi1L-Y, 2022-05-18:
I'm sorry to bother you again. 

Some devices cannot be connected via nRF Connect app for android, Should I get specific equipment ? Or What equipment is allowed to be connected through nRF connect app ?

Thank you !

Wi1L-Y, 2022-05-18:
I am sorry to bother you again.

Before I didn't connect the target through nRF connect. And this time, I try to sniff the target via nRF52 sniffer (selecting target address through wireshark toolbar) and connect the target via nRF Connect app for android in the same time. However, the sniffer cannot get the packets after connection. I find someone else who did it successfully on the website .Do this happen because of the target ?

🙏

Matheus-Garbelini, 2022-05-18:
@Wi1L-Y The steps you described are correct. So indeed it could be your target, although I can't imagine why the sniffer is not able to capture the connection. Perhaps try with a different target.
By the way the advertisements sent by your target are connectable advertisements or non-connectable advertisemetns?

Wi1L-Y, 2022-05-19:
Anyway, thank you very much for your guidance. After changing my target, I get some ACL packets and I will try harder to do more research.
Thank you ! Best

Matheus-Garbelini, 2022-05-19:
@Wi1L-Y glad to know.

closed 2022-05-19 by Matheus-Garbelini