Matheus-Garbelini/sweyntooth_bluetooth_low_energy_attacks #30: Truncated L2CAP attack misalignment between script and paper

mantegna10, 2023-02-07:
According to the White Paper: 

> If the total length of the packet (i.e. LL Length) has
> a value lower than L2CAP Length + 4 for a valid payload,
> then the truncated bytes are copied beyond the underlying
> reception buffer.

In the example (fig. 6 of the white paper) the L2CAP Length is set to 3, while LL Length is 5 (5 < 3+4) and the L2CAP reception buffer is overflown by two bytes (i.e. L2CAP Length+4−LL Length)

Trying to execute:
_DA14580_exploit_att_crash.py_
at **line 139**, the bytes to send are set to:
`length_req = BTLE('7083329a02070000040010edea874aac'.decode('hex'))  # att`

In the packet above the **L2CAP** Length is set to **0** and **LL Length** to **7** (LL Length > L2CAP Length + 4). This is not in accordance with what stated in the paper (See Wireshark capture in the attachment).
[image]

I've tried to change the script, but I'm not able to calculate the CRC.
Thanks for the support.

Matheus-Garbelini, 2023-02-07:
Hi @mantegna10 I believe there was some swap by mistake on the scripts name for variants of this attack.
You can see the truncated l2cap here: https://github.com/Matheus-Garbelini/sweyntooth_bluetooth_low_energy_attacks/blob/master/extras/DA14680_truncated_l2cap_crash.py

alereds94, 2023-02-08:
Hi @Matheus-Garbelini, I have experienced the same issue.

I think the source problem is some mismatch on the table:
https://github.com/Matheus-Garbelini/sweyntooth_bluetooth_low_energy_attacks#available-ble-exploits

where the Truncated L2CAP ([CVE-2019-17517](https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2019-17517)), whose theoretical fundament is explained in Section 6.3 of the white paper, is associated with the script `DA14580_exploit_att_crash.py`.