Matheus-Garbelini/sweyntooth_bluetooth_low_energy_attacks #32: ImportError: No module named html.parser

jsmif, 2023-05-04:
Testing on a raspberry pi 4b running raspbian, with nRF dongle attached.

```
pi@pi1:~/Downloads/sweyntooth_bluetooth_low_energy_attacks $ sudo apt-get install python2.7
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python2.7 is already the newest version (2.7.16-2+deb10u1).
python2.7 set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 398 not upgraded.
```

```
pi@pi1:~/Downloads/sweyntooth_bluetooth_low_energy_attacks $ sudo pip install -r requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pyserial in /usr/lib/python2.7/dist-packages (from -r requirements.txt (line 1)) (3.4)
Collecting nrfutil (from -r requirements.txt (line 2))
  Using cached https://files.pythonhosted.org/packages/95/bd/c69458ec1b9f66b5874bcf7df60f4e768b5fa137663637241628bfdea135/nrfutil-5.2.0.tar.gz
Collecting pycryptodome (from -r requirements.txt (line 3))
  Using cached https://files.pythonhosted.org/packages/b8/2e/cf9cfd1ae6429381d3d9c14c8df79d91ae163929972f245a76058ea9d37d/pycryptodome-3.17.tar.gz
Requirement already satisfied: six in /usr/lib/python2.7/dist-packages (from -r requirements.txt (line 4)) (1.12.0)
Collecting behave~=1.0 (from nrfutil->-r requirements.txt (line 2))
  Using cached https://files.pythonhosted.org/packages/a8/6c/ec9169548b6c4cb877aaa6773408ca08ae2a282805b958dbc163cb19822d/behave-1.2.6-py2.py3-none-any.whl
Requirement already satisfied: click~=7.0 in /usr/lib/python2.7/dist-packages (from nrfutil->-r requirements.txt (line 2)) (7.0)
Collecting crcmod~=1.7 (from nrfutil->-r requirements.txt (line 2))
  Using cached https://files.pythonhosted.org/packages/6b/b0/e595ce2a2527e169c3bcd6c33d2473c1918e0b7f6826a043ca1245dd4e5b/crcmod-1.7.tar.gz
Collecting ecdsa~=0.13.0 (from nrfutil->-r requirements.txt (line 2))
  Using cached https://files.pythonhosted.org/packages/a6/81/2b170b460c84fdc8700cf08aa077ac6a9ff41f4ad3f05d0b3a64ba9f8f2e/ecdsa-0.13.3-py2.py3-none-any.whl
Requirement already satisfied: enum34~=1.0 in /usr/lib/python2.7/dist-packages (from nrfutil->-r requirements.txt (line 2)) (1.1.6)
Collecting intelhex~=2.2 (from nrfutil->-r requirements.txt (line 2))
  Using cached https://files.pythonhosted.org/packages/97/78/79461288da2b13ed0a13deb65c4ad1428acb674b95278fa9abf1cefe62a2/intelhex-2.3.0-py2.py3-none-any.whl
Collecting libusb1~=1.7 (from nrfutil->-r requirements.txt (line 2))
  Using cached https://files.pythonhosted.org/packages/e3/cb/f34ab448d097b5d09b5f6dfc23ddd43bef6f524ed72bd09c9ce6af3ab470/libusb1-1.10.1.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-obi53N/libusb1/setup.py", line 22, in <module>
        from html.parser import HTMLParser
    ImportError: No module named html.parser
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-obi53N/libusb1/
```

jsmif, 2023-05-05:
This seems like it might be related to python2 vs. python3 according to https://stackoverflow.com/questions/23555906/importerror-no-module-named-html-parser ...

However, my pip seems to be using version 2.7

```
pip --version
pip 18.1 from /usr/lib/python2.7/dist-packages/pip (python 2.7)
```

So is the issue that libusb1 is now requiring Python3?

KevynKelso, 2023-07-18:
It seems libusb1 version 1.7 has a compatibility issue with python 2.7. However, for me this was not the case with libusb1==1.7.1.
Try running `pip install libusb1==1.7.1`, then try `pip install nrfutil`