Matheus-Garbelini/sweyntooth_bluetooth_low_energy_attacks #36: NRF52_CMD_CHECKSUM_ERROR

shuimoshusheng, 2024-04-10:
Thank you for your open-source sharing. 
When I use the examples, I always prompt NRF52-CMD-CHECKSUM-ERROR like this：
`root@john:/opt/sweyntooth_bluetooth_low_energy_attacks# python2.7 Telink_zero_ltk_installation.py /dev/ttyACM0 F0:9E:4A:60:B5:DE
Serial port: /dev/ttyACM0
Advertiser Address: F0:9E:4A:60:B5:DE
TX ---> BTLE_ADV / BTLE_SCAN_REQ
Waiting advertisements from F0:9E:4A:60:B5:DE
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
NRF52_CMD_CHECKSUM_ERROR
NRF52_CMD_CHECKSUM_ERROR
NRF52_CMD_CHECKSUM_ERROR
TX ---> BTLE_ADV / BTLE_SCAN_REQ
NRF52_CMD_CHECKSUM_ERROR
TX ---> BTLE_ADV / BTLE_SCAN_REQ
NRF52_CMD_CHECKSUM_ERROR
NRF52_CMD_CHECKSUM_ERROR
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
TX ---> BTLE_ADV / BTLE_SCAN_REQ
NRF52_CMD_CHECKSUM_ERROR
NRF52_CMD_CHECKSUM_ERROR
`
My Device is nRF52480 MDK USB Dongle，Use this method to convert hex to UF2。Then updated firmware[https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/programming/uf2boot/#updating-the-application-firmware](url)

yyh12138, 2024-04-18:
I also meet the same question. It seems that the c in raw_receive() is not correctly resolved.

cyberxml, 2024-05-11:
I was able to successfully program an older nRF52840 MDK.

>UF2 Bootloader 0.2.13-42-g82464f9-dirty lib/nrfx (v2.0.0) lib/tinyusb (legacy-1500-g23df777b) s140 6.1.1
>Model: nRF52840 MDK USB Dongle
>Board-ID: nRF52840-Dongle-v1
>Date: Feb  3 2020

I used the uf2conv tool from the MS github: https://github.com/Microsoft/uf2

`python3 uf2conf.py s140_nrf52_6.1.1_softdevice.hex -c -f 0xADA52840`

And just copied the output (flash.uf2) to the root directory when the MDK was in file mode (hold reset when inserting the usb)

shuimoshusheng, 2024-05-12:
> I was able to successfully program an older nRF52840 MDK.
>
>
> I used the uf2conv tool from the MS github: https://github.com/Microsoft/uf2
>
> `python3 uf2conf.py s140_nrf52_6.1.1_softdevice.hex -c -f [...]

Thanks，I will try it.

closed 2024-10-22 by shuimoshusheng